brocaar/chirpstack-application-server #560: Store uplink is of type confirmed/unconfirmed + dev-addr in postgres integration.

sagar-patel-sls, 2020-11-27:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary
Store uplink is confirmed or not and dev-addr in Postgres integration in `device_up` table of `chirpstack_as_events`

```
create table device_up (
    id uuid primary key,
    received_at timestamp with time zone not null,
    dev_eui bytea not null,
    device_name varchar(100) not null,
    application_id bigint not null,
    application_name varchar(100) not null,
    frequency bigint not null,
    dr smallint not null,
    adr boolean not null,
    f_cnt bigint not null,
    f_port smallint not null,
    tags hstore not null,
    data bytea not null,
    rx_info jsonb not null,
    object jsonb not null,
    confirmed_uplink, boolean not null,
    dev_addr bytea not null
);

-- NOTE: These are example indices, depending on how this table is being
-- used, you might want to change these.
create index idx_device_up_received_at on device_up(received_at);
create index idx_device_up_dev_eui on device_up(dev_eui);
create index idx_device_up_application_id on device_up(application_id);
create index idx_device_up_frequency on device_up(frequency);
create index idx_device_up_dr on device_up(dr);
create index idx_device_up_tags on device_up(tags);
```

## What is the use-case?

- we can easily identify the uplink type in Postgres data

## Can you implement this by yourself and make a pull request?
Yes

brocaar, 2021-02-02:
There is a bit of legacy which I would like to solve, before making changes to the schema of the integration. Unfortunately, the schema of this integration is not automatically managed. I would like to change this first, to make sure that a change to the schema can then be automatically applied, because else it will break existing deployments that use this migration.

To avoid adding more legacy, I propose to use github.com/golang-migrate/migrate/v4/cmd/migrate for schema migrations instead of https://github.com/rubenv/sql-migrate. The latter library pulls in a lot of dependencies. I'm planning to refactor the code that uses sql-migrate in the near future.

One component which already uses the golang-migrate package is https://github.com/brocaar/chirpstack-fuota-server.

Would you be able and willing to implement this as well?

sagar-patel-sls, 2021-02-03:
Hii @brocaar 
Thanks for your answer

Have you decided on any flow or architecture to automatic migration of schema so I can develop that way?

Thanks

brocaar, 2021-02-08:
See the chirpstack-fuota-server repository which already implements schema migrations using golang-migrate. That would be a good start :)

In this case, the first migration must migrate the database in the current schema. I think using `create table if not exists` would be best, as we have to deal with users which already are using the integration, in which case the current schema already exist and with users who have not yet created the tables.

Then the second migration would reflect the changes related to this issue.

sagar-patel-sls, 2021-02-15:
> Would you be able and willing to implement this as well?
I have implemented and created PR.. #573

sagar-patel-sls, 2021-02-23:
> can I make PR for refactor code of sql-migrate?

sagar-patel-sls, 2021-02-23:
Sorry, by mistake it's closed i have reopened

PR #573 (2021-03-01): Implement auto-migration schema of postgresql integration.
Details:
- Auto-migration of schema of the postgres integration
- Used github.com/golang-migrate/migrate/v4/cmd/migrate for schema migrations.

Fixes #560
PR #579 (2021-03-19): Store uplink is of type confirmed/unconfirmed + dev-addr in postgres integration.
Fixes #560 

@brocaar Please review my change.

Is it required to Store `RXINFO` in `JoinEvent`? if required then I will commit to this PR.
closed 2021-03-01 by brocaar