brocaar/chirpstack-application-server #565: Add a new user to an organization API not working - object does not exist

djjoyro, 2020-12-16:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I've got the following error
{
  "error": "object does not exist",
  "code": 5,
  "message": "object does not exist",
  "details": []
}
## What did you expect?
User should be added successfully to the organization
## Steps to reproduce this issue

- Open ChirpStack Application Server REST API
- Authenticate to the server using /api/internal/login endpoint
- Copy jwt token to the JWT TOKEN box
- Expand /api/organizations/{organization_user.organization_id}/users endpoint
- fill organization_user.organization_id with a valid organization id
- fill the body with:
- {
  "organizationUser": {
    "email": "your@email.com",
    "isAdmin": true,
    "isDeviceAdmin": true,
    "isGatewayAdmin": true,
    "organizationID": "<VALID_ORGANIZATIONID>"
  }
}

- Click "Try it out!"

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.13.2  |

sagar-patel-sls, 2020-12-16:
@djjoyro 

I have tested your issue steps..
This is not issue it's already working in `v3.12.2` 

You need to created user first after that you can add to an organization.

1. You need to create one user using POST USER API `/api/users`
2. Try newly created user to an Organization which you want to add

Thanks

djjoyro, 2020-12-16:
Thank you.
It's working correctly. User must be created first using /api/users
After creating, I can assign the user to the company.

closed 2020-12-16 by djjoyro