brocaar/chirpstack-application-server #567: HTTP Integration does not have any timestamp

omogenot, 2021-01-04:
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.
Although an abandoned topic #311 has been closed and relates more or less to that.

## Summary
When JSON data is pushed for Application integration, if the concentrator shield does not have a GPS/GNSS receiver, the { "rxInfo":{"0": {"time":}}} element is left blank, whereas the gateway current time shall be inserted there instead.

## What is the use-case?
A gateway with a concentrator shield having no GPS/GNSS receiver => therefore no GPS time.
However, the ntp protocol is probably accurate enough to provide a valid timestamp.

## Implementation description
If no GPS/GNSS receiver is available, the time field shall be filled using the current date and time (assuming that these values are automatically provided through ntp servers).

## Can you implement this by yourself and make a pull request?
Unfortunately not.

brocaar, 2021-02-02:
> whereas the gateway current time shall be inserted there instead.

See the ChirpStack Gateway Bridge `fake_rx_time` config option.

closed 2021-02-02 by brocaar

omogenot, 2021-02-02:
This option is not included in the default gateway bridge config file in the Chirpstack OS version.
It might be a good idea to add it.

brocaar, 2021-02-04:
@omogenot I'm trying to keep the default config as simple as possible, but please note the comment on the top:
https://github.com/brocaar/chirpstack-gateway-os/blob/master/layers/chirpstack/meta-chirpstack/recipes-chirpstack/chirpstack-gateway-bridge/chirpstack-gateway-bridge/chirpstack-gateway-bridge.toml#L2

Maybe it would be an idea however to change the https://github.com/brocaar/chirpstack-concentratord behavior, so that the time can actually be set at the source (even if there is no GPS time available).