brocaar/chirpstack-application-server #568: Sporadic uplink error UPLINK_FCNT_RETRANSMISSION

omogenot, 2021-01-05:
- [X] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I have two different setups:

**Setup1**: Original Multitech conduit, connected to a Debian Linux box with ChirpStack-application-server (version 3.13.2)
**Setup2**: Raspberry PI 3+ with IC880A hat using ChirpStack OS (version 3.12.1 = latest available on the site)

When I join a device to the **Setup1**, and I go to "DEVICE DATA" tab of this device, I can see the various frames exchanged  (join, up, status) followed by a list of up frames at regular interval => OK

When I join the same device to **Setup2**, and go to the "DEVICE DATA" tab of the device, I can see a list of up and error frames. The error frames are complaining that the fCnt value did not change, whereas in reality this value is different and moves forward. Therefore, the up frame shall not be discarded as an error. By the way, this erroneous list of error frames creates a lot of data miss in the final application database.
Please note that Raspberry Pi/IC880A device was previously used as an original Semtech packet_forwarder, connected to the **Setup1** server and never had any of such issues. 

## What did you expect?

I expect **Setup2** to work as **Setup1**.

## Steps to reproduce this issue

See above.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.12.1 |
| Network Server      | v3.10.0 |
| Gateway Bridge      | v3.9.2 |
| Chirpstack API      | v1.0.0 |
| Geolocation         |   N/A   |
| Concentratord       | v3.0.2 |

brocaar, 2021-02-02:
This error is to make you aware that your device re-used the uplink fCnt. This might be a re-transmission or replay attack. With LoRaWAN 1.0.x, the NS can not know the difference therefore if will not continue the processing of an uplink when it has already seen the uplink fCnt.

I don't think this has anything to do between setup1 and setup2.

omogenot, 2021-02-02:
Sorry to insist, however the two setup so behave differently. The device that had these errors (about 33% of frames) on Setup2 has been re-assigned to Setup1 and no error occurred.
Please note also in the error description that the error is shown although the uplink fCnt is NOT the same (it is incremented) and the original frame is not displayed. If it were a re-transmission or attack, the original frame should be displayed first.

omogenot, 2021-02-08:
I just noticed that the ntpd daemon seems not to work correctly, therefore the raspberry pi os date and time is incorrect (3 days behind in my case), and it seems that the error rate is increasing when the os date and time is wrong.
I have a test setup running to confirm this...
I had to restart the ntpd daemon using `sudo /etc/init.d/ntpd start` and I'll wait to see how long this will work.

closed 2021-06-15 by brocaar