brocaar/chirpstack-application-server #571: Authentication via X headers like "X-Remote-Email" from Authelia

andryyy, 2021-01-27:
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

## What is the use-case?

Using tools like Authelia which pass X headers to the application after authenticating against them.

## Implementation description

Chirpstack could try to read the X-Remote-Email header as set by the proxy and try to determine an existing user with that address.

## Can you implement this by yourself and make a pull request?

Sadly no. :-(

andryyy, 2021-02-24:
@brocaar Is that something I could sponsor for the hourly rate you sent me a while ago?

brocaar, 2021-03-22:
Hello @andryyy, I do see that there could be use-cases for this feature, but currently there are a few other items I would like to prioritize first, before I could start working on this.

closed 2025-05-02 by brocaar