brocaar/chirpstack-application-server #572: /api/applications returns empty result list, but "totalCount" says that there should be items

SebastianKotstein, 2021-02-04:
- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
When trying to fetch the list of applications via the HTTP API of the application server, the result list is empty, but the "totalCount" property (as well as the web interface of the application server) indicates that there must be, in my setup, 10 applications. 

## What did you expect?
A JSON object for each application contained in "result"
## Steps to reproduce this issue

Steps:

1. curl -X GET --header 'Accept: application/json' --header 'Grpc-Metadata-Authorization: Bearer <TOKEN>  'http://MY_IP/api/applications'
2. Response is:
`{
    "totalCount": "10",
    "result": []
}`

## Could you share your log output?

```shell
INFO[0000] starting ChirpStack Application Server        docs="https://www.chirpstack.io/" version=3.14.0
INFO[0000] storage: setting up storage package
INFO[0000] storage: setup metrics
INFO[0000] storage: setting up Redis client
INFO[0000] storage: connecting to PostgreSQL database
INFO[0000] storage: applying PostgreSQL data migrations
INFO[0000] storage: PostgreSQL data migrations applied   count=0
INFO[0000] integration: configuring global integrations
INFO[0000] integration/mqtt: TLS config is empty
INFO[0000] integration/mqtt: connecting to mqtt broker   server="tcp://localhost:1883"
INFO[0000] api/as: starting application-server api       bind="0.0.0.0:8001" ca_cert= tls_cert= tls_key=
INFO[0000] integration/mqtt: connected to mqtt broker
INFO[0000] integration/mqtt: subscribing to tx topic     qos=0 topic=application/+/device/+/command/down
FATA[0000] setup api error: setup application-server api error: start application-server api listener error: listen tcp 0.0.0.0:8001: bind: addr

sudo netstat -tlnp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      919/systemd-resolve
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1696/sshd
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      1404/postgres
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      1375/mosquitto
tcp        0      0 0.0.0.0:1501            0.0.0.0:*               LISTEN      1604/dsmcad
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN      1160/redis-server 1
tcp6       0      0 :::22                   :::*                    LISTEN      1696/sshd
tcp6       0      0 ::1:5432                :::*                    LISTEN      1404/postgres
tcp6       0      0 :::1883                 :::*                    LISTEN      1375/mosquitto
tcp6       0      0 :::8000                 :::*                    LISTEN      1289/chirpstack-net
tcp6       0      0 :::8001                 :::*                    LISTEN      1263/chirpstack-app
tcp6       0      0 :::8003                 :::*                    LISTEN      1263/chirpstack-app
tcp6       0      0 ::1:6379                :::*                    LISTEN      1160/redis-server 1
tcp6       0      0 :::8080                 :::*                    LISTEN      1263/chirpstack-app

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.14.0  |
| Network Server      | v3.12.0  |
| Gateway Bridge      | v3.10.0  |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2021-02-04:
It looks like you didn't specify the `limit` argument in your request, in which case you do get the total count, but not the result-set.

closed 2021-02-04 by brocaar

SebastianKotstein, 2021-02-04:
Many thanks for your quick reply! So it is not an issue as supposed, but a feature. I have tested it again and now it works!