brocaar/chirpstack-application-server #575: Drop of the downstream messages

lexfrei88, 2021-02-12:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
Only one of multiple downstream messages is handled if they will appear in MQTT with a delay of less than 100ms (all messages to the same device)

## What did you expect?
All downstream message handled

## The issue I believe in here [mqtt.go#txPayloadHandler](https://github.com/brocaar/chirpstack-application-server/blob/65b5d15c55062c583ce9253ba124343e7fa36a80/internal/integration/mqtt/mqtt.go#L410):
```
	// Since with MQTT all subscribers will receive the downlink messages sent
	// by the application, the first instance receiving the message must lock it,
	// so that other instances can ignore the message.
	key := fmt.Sprintf("lora:as:downlink:lock:%d:%s", pl.ApplicationID, pl.DevEUI)
	set, err := storage.RedisClient().SetNX(key, "lock", downlinkLockTTL).Result()
	if err != nil {
		log.WithError(err).Error("integration/mqtt: acquire lock error")
		return
	}
```
Two messages will be considered different only if they send to different devices. Why this code can't take into account the content of these messages? That would help to avoid duplications but make it possible to process different messages that came too often

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.10.0 |

brocaar, 2021-02-15:
Yes, I agree. Would you like to create a PR for this?

lexfrei88, 2021-02-21:
Done. [The link.](https://github.com/brocaar/chirpstack-application-server/pull/577)

commit 3dc8ff33 (2021-02-24): Base MQTT downlink lock key on payload content. (#577)

Closes #575
closed 2021-02-24 by brocaar