brocaar/chirpstack-application-server #578: "panic: runtime error: invalid memory address or nil pointer dereference" while adding tls certificates in web interface

kannansks, 2021-03-03:
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: time="2021-03-03T00:09:28-05:00" level=info msg="creating network-server client" server="10.10.20.58:8000"
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: panic: runtime error: invalid memory address or nil pointer dereference
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x60 pc=0xcc0106]
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: goroutine 849 [running]:
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-application-server/internal/storage.CreateNetworkServer(0x17be5a0, 0xc000770c00, 0x7ff5c0467d80, 0xc00000e318, 0xc0001eddc0, 0x40ccd8, 0x8)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /chirpstack-application-server/internal/storage/network_server.go:108 +0x906
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-application-server/internal/api/external.(*NetworkServerAPI).Create.func1(0x17cafc0, 0xc00000e318, 0x0, 0x0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /chirpstack-application-server/internal/api/external/network_server.go:58 +0x96
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-application-server/internal/storage.Transaction(0xc0002e72c0, 0xc0001eddc0, 0xc000770c00)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /chirpstack-application-server/internal/storage/db.go:129 +0xa9
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-application-server/internal/api/external.(*NetworkServerAPI).Create(0xc00047cf80, 0x17be5a0, 0xc000770c00, 0xc000770a80, 0xc00047cf80, 0x7ff5c061c0c8, 0xc00079ebc0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /chirpstack-application-server/internal/api/external/network_server.go:57 +0x35b
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-api/go/v3/as/external/api._NetworkServerService_Create_Handler.func1(0x17be5a0, 0xc000770c00, 0x1383d40, 0xc000770a80, 0x20, 0xc0009693b0, 0xc0006e3478, 0xaa4fd3)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/brocaar/chirpstack-api/go/v3@v3.7.8/as/external/api/networkServer.pb.go:846 +0x86
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-prometheus.(*ServerMetrics).UnaryServerInterceptor.func1(0x17be5a0, 0xc000770c00, 0x1383d40, 0xc000770a80, 0xc0001595e0, 0xc000159600, 0xc0002137e8, 0xc0006e3578, 0x333738352d373032, 0x3338383065353937)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-prometheus@v1.2.0/server_metrics.go:107 +0xad
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1.1.1(0x17be5a0, 0xc000770c00, 0x1383d40, 0xc000770a80, 0x0, 0xc00014c140, 0x17be5a0, 0xc000770c00)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/chain.go:25 +0x63
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-application-server/internal/logging.UnaryServerCtxIDInterceptor(0x17be5a0, 0xc000770c00, 0x1383d40, 0xc000770a80, 0xc0001595e0, 0xc000159620, 0x0, 0x0, 0x0, 0x0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /chirpstack-application-server/internal/logging/logging.go:50 +0x2fd
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1.1.1(0x17be5a0, 0xc000770bd0, 0x1383d40, 0xc000770a80, 0x20, 0xc007e46213a0c174, 0xd17d49a54, 0x26a8500)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/chain.go:25 +0x63
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware/logging/logrus.UnaryServerInterceptor.func1(0x17be5a0, 0xc000770ae0, 0x1383d40, 0xc000770a80, 0xc0001595e0, 0xc000159640, 0x17be5a0, 0xc000770ae0, 0xc00000e2e8, 0xc000770ab0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/logging/logrus/server_interceptors.go:32 +0x110
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1.1.1(0x17be5a0, 0xc000770ae0, 0x1383d40, 0xc000770a80, 0x20, 0x1383d40, 0xc000770a80, 0x0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/chain.go:25 +0x63
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware/tags.UnaryServerInterceptor.func1(0x17be5a0, 0xc000770a50, 0x1383d40, 0xc000770a80, 0xc0001595e0, 0xc000159660, 0xb3784a, 0x1341220, 0xc000159680, 0xc0001595e0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/tags/interceptors.go:22 +0x86
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1.1.1(0x17be5a0, 0xc000770a50, 0x1383d40, 0xc000770a80, 0xc000415600, 0x0, 0xc0006e3b18, 0x40ccd8)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/chain.go:25 +0x63
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1(0x17be5a0, 0xc000770a50, 0x1383d40, 0xc000770a80, 0xc0001595e0, 0xc000159600, 0xc00080ab88, 0x4f8558, 0x1370bc0, 0xc000770a50)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/grpc-ecosystem/go-grpc-middleware@v1.1.0/chain.go:34 +0xd5
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: github.com/brocaar/chirpstack-api/go/v3/as/external/api._NetworkServerService_Create_Handler(0x13011a0, 0xc00047cf80, 0x17be5a0, 0xc000770a50, 0xc000577b60, 0xc00074b080, 0x17be5a0, 0xc000770a50, 0xc00016a000, 0x1f46)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/github.com/brocaar/chirpstack-api/go/v3@v3.7.8/as/external/api/networkServer.pb.go:848 +0x14b
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: google.golang.org/grpc.(*Server).processUnaryRPC(0xc0001a4d00, 0x17ceda0, 0xc0009677a0, 0xc000415600, 0xc00074b650, 0x2696260, 0x0, 0x0, 0x0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/google.golang.org/grpc@v1.28.0/server.go:1082 +0x50a
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: google.golang.org/grpc.(*Server).handleStream(0xc0001a4d00, 0x17ceda0, 0xc0009677a0, 0xc000415600, 0x0)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/google.golang.org/grpc@v1.28.0/server.go:1405 +0xcc9
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: google.golang.org/grpc.(*Server).serveStreams.func1.1(0xc00014c0f0, 0xc0001a4d00, 0x17ceda0, 0xc0009677a0, 0xc000415600)
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/google.golang.org/grpc@v1.28.0/server.go:746 +0xa1
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]: created by google.golang.org/grpc.(*Server).serveStreams.func1
Mar 03 00:09:28 localhost.localdomain chirpstack-application-server[2005]:         /go/pkg/mod/google.golang.org/grpc@v1.28.0/server.go:744 +0xa1
Mar 03 00:09:28 localhost.localdomain systemd[1]: chirpstack-application-server.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 03 00:09:28 localhost.localdomain systemd[1]: chirpstack-application-server.service: Failed with result 'exit-code'.

@brocaar  this is the application server journal when i try to update network server with tls certificates.

brocaar, 2021-03-31:
My assumption is that this is with the latest version, is that correct? Please use the provided template the next time when reporting an issue, e.g. it asks about the versions which you are using + the exact steps to re-produce the issue.

closed 2021-03-31 by brocaar

kannansks, 2021-03-31:
okay @brocaar  will stick with the template next time.
I am currently using application server v3.13.2 and the issue occurred when i tried adding tls certificates but the error stopped when i changed the config file of certificates.

brocaar, 2021-04-01:
@kannansks the above commit should fix the issue :)