brocaar/chirpstack-application-server #586: app server fails to compile  bcs of //go:embed directive in static.go file

Muslimah94, 2021-04-19:
- [ ] The issue is present in the latest release.

- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I couldn't start chirpstack application server from source code
## What did you expect?
Chirpstack Application Server successfully running on localhost 
## Steps to reproduce this issue

Steps:

1. To clone the latest version of app server or any other version starting from [https://github.com/brocaar/chirpstack-application-server/commit/aa580165febe3fd84c22d2da3cae97706e5a3a38](Refactor SQL schema migrations from sql-migrate to golang-migrate + U…)
2. To execute `make dev-requirements ui-requirements` command in the app server directory 
3. To execute `make build` command in the app server directory 
4. Face an error which says smth like:
`static/static.go:6:75: pattern *.html: no matching files found
make: *** [Makefile:7: build] Error 1`
or 
`static/static.go:6:75: pattern *.js: no matching files found
make: *** [Makefile:7: build] Error 1`

## Could you share your log output?

```shell$ make ui-requirements 
Installing UI requirements
up to date, audited 2121 packages in 8s

129 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
$ make build
Building ui

> chirpstack-application-server-ui@3.0.0 build
> react-scripts build

Creating an optimized production build...
Compiled successfully.

File sizes after gzip:

  470.45 KB  build/static/js/2.c34da569.chunk.js
  52.03 KB   build/static/js/main.97a28eae.chunk.js
  23.56 KB   build/static/css/2.e72d3014.chunk.css
  791 B      build/static/js/runtime-main.33db2bcb.js
  116 B      build/static/css/main.3f9ef032.chunk.css

The project was built assuming it is hosted at /.
You can control this with the homepage field in your package.json.

The build folder is ready to be deployed.
You may serve it with a static server:

  npm install -g serve
  serve -s build

Find out more about deployment here:

  https://cra.link/deployment

Fetching Swagger definitions and generate combined Swagger JSON
Клонирование в «/tmp/chirpstack-api»…
remote: Enumerating objects: 3777, done.
remote: Counting objects: 100% (171/171), done.
remote: Compressing objects: 100% (133/133), done.
remote: Total 3777 (delta 78), reused 94 (delta 31), pack-reused 3606
Получение объектов: 100% (3777/3777), 2.68 MiB | 319.00 KiB/s, готово.
Определение изменений: 100% (2406/2406), готово.
Note: switching to 'v3.9.5'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD сейчас на a49efdc Add ReEncryptDeviceQueueItems method on AS interface.
mkdir -p build
go build  -ldflags "-s -w -X main.version=3.15.0-test.1-1-g5384962" -o build/chirpstack-application-server cmd/chirpstack-application-server/main.go
static/static.go:6:82: pattern *.js: no matching files found
make: *** [Makefile:7: build] Ошибка 1
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.14.0  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

sagar-patel-sls, 2021-04-20:
@Muslimah94 
i have clone `chirpstack-application-server` and perform the build step and it's working.

```
$ make build
Building ui

> chirpstack-application-server-ui@3.0.0 build E:\chirpstack-application-server\ui
> react-scripts build

Creating an optimized production build...
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Compiled successfully.

File sizes after gzip:

  454.24 KB  build\static\js\2.7e037a47.chunk.js
  51.08 KB   build\static\js\main.27461f3b.chunk.js
  23.57 KB   build\static\css\2.ff340785.chunk.css
  789 B      build\static\js\runtime-main.ce245b8e.js
  116 B      build\static\css\main.a2b7b0b6.chunk.css

The project was built assuming it is hosted at /.
You can control this with the homepage field in your package.json.

The build folder is ready to be deployed.
You may serve it with a static server:

  npm install -g serve
  serve -s build

Find out more about deployment here:

  bit.ly/CRA-deploy

Fetching Swagger definitions and generate combined Swagger JSON
Cloning into '/tmp/chirpstack-api'...
remote: Enumerating objects: 3777, done.
remote: Counting objects: 100% (171/171), done.
remote: Compressing objects: 100% (133/133), done.
remote: Total 3777 (delta 78), reused 94 (delta 31), pack-reused 3606
Receiving objects: 100% (3777/3777), 2.68 MiB | 975.00 KiB/s, done.
Resolving deltas: 100% (2406/2406), done.
Note: switching to 'v3.9.5'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at a49efdc Add ReEncryptDeviceQueueItems method on AS interface.
mkdir -p build
go build  -ldflags "-s -w -X main.version=3.15.0-test.1-2-g6eb051c" -o build/chirpstack-application-server cmd/chirpstack-application-server/main.go
```

wereii, 2021-04-21:
I am stuck on the same problem, doing `make build` fails at the go:embed 
```sh
mkdir -p build
go build  -ldflags "-s -w -X main.version=3.15.0-test.1-3-g2eab64c" -o build/chirpstack-application-server cmd/chirpstack-application-server/main.go
static/static.go:6:75: pattern *.html: no matching files found
make: *** [Makefile:7: build] Error 1
```

go version go1.16.3 linux/amd64

@sagar-patel-sls Could you provide your `go version` output?

brocaar, 2021-04-21:
Are you using the Docker Compose based environment? It is the recommended way of building the project as it comes with the dev. environment pre-setup for you.

wereii, 2021-04-21:
This was outside of docker.

I don't know why but for me both `docker-compose build` and `docker-compose up` fails at the last Dockerfile line `RUN make dev-requirements` - I do have a go setup that needs forked chirpstack-api with `replace` so I uncommented the respective volume yet unless I do `docker-compose exec chirpstack-application-server` the volume'd `/chirpstack-api` folder is empty.
I spent few hours on it now and I still have no clue where this bug comes from, if it's some docker miss-configuration on my end, something overlooked in dockerfile, or who knows.

Anyways, this issue does not occur for me in that docker.

Muslimah94, 2021-04-22:
> I am stuck on the same problem, doing `make build` fails at the go:embed
> 
> ```shell
> mkdir -p build
> go build  -ldflags "-s -w -X main.version=3.15.0-test.1-3-g2eab64c" -o build/chirpstack-application-server cmd/chirpstack-application-server/main.go
> static/static.go:6:75: pattern *.html: no matching files found
> make: *** [Makefile:7: build] Error 1
> ```
> 
> go version go1.16.3 linux/amd64
> 
> ~@sagar-patel-sls Could you provide your `go version` output?~
> 
> Edit:
> It seems that somehow the `ui-requirements` step failed for me so the static folder didn't get populated properly.
> Doing something along these commands `make clean && rm -rf ui/node_modules ui/package-lock.json && make dev-requirements ui-requirements` resolved the `go:embed` problem for me.

Luckily, this solution worked for me just fine, thanks a lot!

closed 2021-04-22 by Muslimah94