brocaar/chirpstack-application-server #600: Send `fCtrl.macPayload.ack` in UplinkEvents to integrations

narrowtux, 2021-05-17:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

in UpEvent messages to integrations, the ack flag that is visible in the frame log of the UI should also be transmitted via the UplinkEvent to integrations, such as the HTTP Webhook integration. Currently, there's no such field documented and also none transmitted that looks like it would be the `ack` flag.

## What is the use-case?

I want to be able to send confirmed down messages with class C and then use the next up frame from the device to check if it has actually received the message.

## Implementation description

Either loop the entire macPayload object into the UplinkEvent, or just `ack` flag.

(https://github.com/brocaar/chirpstack-api/blob/master/protobuf/as/integration/integration.proto#L48)

## Can you implement this by yourself and make a pull request?

I'm not fluent in go

narrowtux, 2021-05-17:
Maybe this is actually a bug in the AckEvent (https://github.com/brocaar/chirpstack-api/blob/master/protobuf/as/integration/integration.proto#L130), which, if I understand correctly, should be pushed with `acknowledged: true` whenever the device sends an upframe with ack.

Currently, with a class C device, the behaviour is like this:

 * my app posts a new confirmed down frame to a class C device (OTAA mac v1.0)
 * chirpstack returns the frame counter in that request
 * chirpstack sends a TxAckEvent after it has pushed the down frame to the gateway
 * immediately after that, chirpstack sends an AckEvent with acknowledge: false
 * next time the device sends an up frame, no AckEvent with acknowledge: true is sent.

brocaar, 2021-05-17:
> immediately after that, chirpstack sends an AckEvent with acknowledged: false

This looks like a misconfiguration in the device-profile. Please check the Class-C timeout value. ChirpStack will wait up to that timeout until it considers the downlink unacknowledged (if no ack was received by then). if an uplink with `ack: true` is received before the timeout occurs, you will receive an `ack` event with acknowledged `true` instead of `false`.

closed 2021-05-17 by brocaar

narrowtux, 2021-05-17:
I see, that did the trick, thank you.