brocaar/chirpstack-application-server #604: Chirpstack stops sending data over kafka after disconnect.

thisIsAnil, 2021-06-15:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I'm using kafka integration in SASL_PLAINTEXT mode with chirpstack running as docker container. I observed when kafka goes down and is back again chirpstack does not resend the data which arrived when kafka was down. Additionally after kafka is up again chirpstack does not send new arriving data as well until I restart chirpstack server. I did not see any error in logs with debug mode enabled.
## What did you expect?
Chirpstack should resend data packets which arrived when kafka was down when using ACK=1 or ACK=all. Default kafka config is ACK=1 and 2147483647 times retries. Additionally I suspect chirpstack does not reconnect to kafka again after it was down but the default kafka producer config supports reconnecting. 
## Steps to reproduce this issue

Steps:

1. Setup up Kakfa integration with chirpstack docker env.
2. Allow chirpstack to connect to kafka and send some data.
3. Stop kafka and wait for some time such that data from some device arrives ( may > 20 mins)
4. Start kafka again.

## Could you share your log output?
Chirpstack application server logs show no error even in debug mode and logs following message irrespective of if kafka was connected or not.

> time="2021-06-15T05:15:38Z" level=info msg="integration/kafka: publishing message" ctx_id=c678690d-ecbf-40f1-9110-ad4473f3b8b9 key=application.${appId}.device.${deviceEUI}.event.up

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.14.0  |
| Network Server      |  v3.12.1  |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2021-06-15:
I believe it should re-connect automatically, this is the used Kafka client: https://github.com/segmentio/kafka-go#writer-

> Automatic retries and reconnections on errors.

Please note that most libraries do implement a back off in their re-connect logic. E.g. while a re-connect might happen initially after a second, this interval might increase over time to potentially minutes. Are you sure this is not the case?

thisIsAnil, 2021-07-05:
Update: 
The 0.4.17 release of segmentio  kafka-go library fixes issues mentioned here. @brocaar Can you bump up the kafka-go version in chirpstack?
Complete info about release can be found [here](https://github.com/segmentio/kafka-go/releases/tag/v0.4.17)

commit d9854676 (2021-07-07): Update kafka client library.

Closes #604.
closed 2021-07-07 by brocaar