brocaar/chirpstack-application-server #622: Revert Fix log duplication on WebSocket re-connect.

sagar-patel-sls, 2021-09-13:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
We need to revert `log duplication on WebSocket re-connect` [0518d48](https://github.com/brocaar/chirpstack-application-server/commit/0518d4851fabad0a5a40ae39c1a13ddbf88adc65) commit.

- I have checked and tested `Live frames and event logs` features with default 10 records.
- Before this commit [0518d48](https://github.com/brocaar/chirpstack-application-server/commit/0518d4851fabad0a5a40ae39c1a13ddbf88adc65), it was working fine after this change, some of the uplinks or redis stream records are not showing on the web interface. 
- Redis streams show packets but not display in the web interface.

## What did you expect?
All Redis streams records show on the web interface.

## Could you share your log output?

- I have debugged this issue as I have found the below issue in the web interface.
- Redis Stream shows `9` records saved and the web interface shows only `7` records.

1. DeviceData.js
`if (data.length === 0 || moment(d.publishedAt).isAfter(data[0].publishedAt)) {}`

I have debugged using `console.log(data[0].publishedAt)` and shown `undefined`.
[image]

2. DeviceFrames.js

`if (frames.length === 0 || moment(f.publishedAt).isAfter(moment(frames[0].publishedAt))) {}`

I have debugged using `console.log(frames[0].publishedAt shows undefined)` and shown `undefined`.
[image]

3. GatewayFrames.js
`if (frames.length === 0 || moment(f.publishedAt).isAfter(moment(frames[0].publishedAt))) {}`

I have debugged using `console.log(frames[0].publishedAt shows undefined)` and shown `undefined`.

**Redis Stream logs (Stored `9` records)**
[image]

**Device Data (Shows only `7` records)**
[image]

## Resolutions
When I have debugged this issue with the comment code of this commit [0518d48](https://github.com/brocaar/chirpstack-application-server/commit/0518d4851fabad0a5a40ae39c1a13ddbf88adc65) and checked the result, it's working fine. 
```javascript
onData(d) {
    if (this.state.paused) {
      return;
    }

    let data = this.state.data;
    const now = new Date();

    //console.log(data[0].publishedAt);
    //if (data.length === 0 || moment(d.publishedAt).isAfter(data[0].publishedAt)) {
      data.unshift({
        id: now.getTime(),
        publishedAt: d.publishedAt,
        type: d.type,
        payload: JSON.parse(d.payloadJSON),
      });

      this.setState({
        data: data,
      });
    //}
  }
```

[image]

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.17.0 |
| Network Server      | v3.15.0 |
| Gateway Bridge      |v3.13.1|

brocaar, 2021-09-20:
> console.log(data[0].publishedAt);

Did you add this line? Then I think this is the issue. Please see the line below:

> if (data.length === 0 || moment(d.publishedAt).isAfter(data[0].publishedAt)) {

`data.length` could be `0`, thus only when `data.lenth > 0` you could take the first item using `data[0]`. That is why you are seeing the `undefined` error.

Also I don't think we should uncomment this line:

> if (data.length === 0 || moment(d.publishedAt).isAfter(data[0].publishedAt)) {

Its purpose is to (entries) to the overview under two conditions:

* `data.length === 0`, thus this is the first item
* If the `publishedAt` timestamp is after the previous item

It is there to avoid duplicated entries on websocket reconnect as both conditions would fail. Please let me know if I'm overlooking something.

sagar-patel-sls, 2021-09-20:
> Did you add this line? Then I think this is the issue. Please see the line below:

Yes, I have added this `console.log(data[0].publishedAt);` line and tested but I got the above error.

I have checked and debug the above two conditions but it fails and getting the above errors. I think we need to think of something different.

Thanks

brocaar, 2021-09-20:
But please see your first screenshot, it fails on:

`console.log(data[0].publishedAt);`

Which can be expected as `data` is an array and there is no guarantee that it contains items, you are requesting an index that does not exist. You should remove that line first before you can debug this further.

sagar-patel-sls, 2021-09-20:
Yes, I have removed the below line and debug but not getting success

`console.log(data[0].publishedAt);`

I have added the above line for testing purposes.

1. I have tested without the above line but some of the redis streams not showing on the Device data tab (see section 3: GatewayFrames.js).
2. After this issue I have removed [0518d48](https://github.com/brocaar/chirpstack-application-server/commit/0518d4851fabad0a5a40ae39c1a13ddbf88adc65) this commit code and it's working fine.
3. For debugging purposes I have added console logs to check `publishedAt` value in the `data` array that's why I have added the below line.
`console.log(data[0].publishedAt);`

4. I have verified this issue with the help of the `redis-cli` and `device data` tab (See Screenshot: Redis Stream logs (Stored 9 records) & Device Data (Shows only 7 records))
5. I have attached the first screenshot for your reference.

if you need more information please let me know...
Thanks

sagar-patel-sls, 2021-09-29:
@brocaar 

Any update?

sagar-patel-sls, 2021-10-06:
Hi @brocaar 

This issue is resolved by this [651b572](https://github.com/brocaar/chirpstack-application-server/commit/651b57253b41fb7f653f93cc7872f07740e41b6c) commit.

closed 2021-10-06 by brocaar