brocaar/chirpstack-application-server #651: how get version on api request

geniv, 2021-12-20:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

How get number version from API? 

## What is the use-case?

Problem is between version 3.15 and 3.17. And binary of application server for argument "version" returm empty string: "".
This BC is totaly bad for my driver, when Idont know which is actual version of application server.

## Implementation description

Any chance create endpoint eg.: "GET internal/version" or "GET me/version" where return string will be text of current version.

## Can you implement this by yourself and make a pull request?

Yes, only in php, maybe java :)

geniv, 2021-12-29:
@brocaar command `# /usr/bin/chirpstack-application-server version` or `$ /usr/bin/chirpstack-application-server version` on terminal in docker return "" string (empty string), but version 3.14.0 or 3.15.0 is function correct and return string with current version.

liangyuanpeng, 2021-12-30:
I can get the correct info from docker container.  

```
~ # ./lora-app-server version
3.4.0
```  

Also success on my terminal.   
```
lan@lan:~/repo/git/chirpstack-application-server$ build/chirpstack-application-server version
WARN[0000] No configuration file found, using defaults. See: https://www.chirpstack.io/application-server/install/config/ 
3.17.3-12-gb911e629
```  

I agree with this feature and i can implement if the feature accepted.

geniv, 2021-12-30:
> I can get the correct info from docker container.

yes, I know, it is work in version 3.15, but in latest version application server 3.17.4 this command return just empty string

brocaar, 2022-01-05:
I thought I had solved this, but it still seems present. The issue seems to be the migration to GitHub actions. It seems that the `.git` directory is not provided to the compilation step within the Docker environment.