brocaar/chirpstack-application-server #652: Feature: Support other script engine besides Otto

chopmann, 2021-12-23:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

## What is the use-case?
The JS-Engine "Otto" while being providing a solid foundation, is lacking on some of the more modern ways/features of writing JS.
@sagar-patel-sls mentioned goja (https://github.com/dop251/goja)
something like yaegi (https://github.com/traefik/yaegi) could also be useful, and provide a standard set of utils.

## Implementation description

Forum User: https://forum.chirpstack.io/u/marsman
> To make the replacement of the JavaScript engine not breaking anything it seems to be an option to extend the DropDown list to choose the JavaScript engine (current “Custom JavaScript codec functions” should be renamed to indicate that his refers to the outdated Otto JS engine).
On updates this can would still default to the Otto JS Engine, so the old scripts still work and if you add a new “Device profile” the Codec would use the new JavaScript engine (=default to goja)

## Can you implement this by yourself and make a pull request?

Some parts of it. Depends which engine(s) we want to add/support.

brocaar, 2021-12-23:
I'm aware of its limitations. While you will not find anything on GitHub yet, I am working on a refactor and cleanup of the code-base. One of the items that this refactor will address is replacing the Otto JS engine. I'm planning to write something on the forum about these changes soon.

hubpav, 2022-01-31:
@brocaar Are you planning to abandon Otto whatsoever in the future? If that is the case, this feature request I have submitted would have a short lifespan: https://github.com/brocaar/chirpstack-application-server/issues/664