brocaar/chirpstack-application-server #664: Upgrade Otto Javascript interpreter

hubpav, 2022-01-31:
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

Please, update Otto Javascript interpreter to the most recent version with the next ChirpStack AS release. Currently, the repo is locked against this commit:

https://github.com/robertkrimen/otto/tree/37f8e9a2460cc10ee975f900b11af0fc55ca1406

## What is the use-case?

There have been numerous bug fixes in Otto since Dec 17, 2019. I came across expression evaluation incorrectness as compared to Node.js. Unfortunately, I did not have time to test against the most recent version of Otto's most recent commit, but given the number of bugs fixed in the meantime, the preemptive update to Chirpstack should be done anyway.

Thanks!

hubpav, 2022-05-10:
@brocaar Any update on this, please?

brocaar, 2022-05-10:
@hubpav please see https://forum.chirpstack.io/t/help-testing-chirpstack-v4-test-releases/13970. ChirpStack v4 contains a different JS runtime and will also include the NodeJS "Buffer" class (e.g. for base64 or hex encoding / decoding).

I'm happy to review a PR to upgrade Otto for the current v3 version.

hubpav, 2022-05-10:
@brocaar Oh, wow! Having the Node.js Buffer class available is a game-changer for us :) No point to upgrade v3 then! P.

brocaar, 2022-05-10:
Please note that the above test-releases do not include the Buffer class yet, I pushed this change literally two hours ago :-) (https://github.com/chirpstack/chirpstack/commit/5b1d70cdbfc99a195a6e162d8bfbf8df8b17a7ec)

hubpav, 2022-05-10:
@brocaar Alright - will wait for the next test release. BTW Is there an estimated milestone for the stable v4 release?

closed 2025-05-02 by brocaar