brocaar/chirpstack-application-server #665: Add new user failed

przemekgalazka, 2022-02-03:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
when adding new user we get
**object does not exist (code: 5)**

## What did you expect?
new user added

## Steps to reproduce this issue

Steps:

1. Open /#/organizations/1/users/create
2. Write username
3. hit add button
4. and you get error

## Could you share your log output?
chirpstack-application-server_1  | time="2022-02-03T14:36:25.274903456Z" level=info msg="finished unary call with code NotFound" ctx_id=e92f8429-4f96-4057-8527-aaf52be1cdb3 error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=AddUser grpc.service=api.OrganizationService grpc.start_time="2022-02-03T14:36:25Z" grpc.time_ms=1.879 peer.address="127.0.0.1:50956" span.kind=server system=grpc

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.17.4  |

mfilotto, 2022-04-15:
Any update ?
I have the same issue when creating a user under an Org

brocaar, 2022-04-15:
Looking at the error (object does not exist), it looks like you are trying to add a user that doesn't exist. Note that adding a user to an organization only works for users that already exist in the database. It does not create new users.

closed 2022-04-15 by brocaar

mfilotto, 2022-04-18:
It's working just fine when user is added globally first 
Thanks for your answer