brocaar/chirpstack-application-server #673: chirpstack-application-server  clear device information ！！！

zgz757183190, 2022-03-09:
Hi  God, 
    under what circumstances will the chirpstack-application-server empty the "Device address" and "Device address" and "Application session key" of the device.

Recently, this has always happened in my project, but I can't find the reason.

AS   3.8
NS   3.8 
GW 3.8

liangyuanpeng, 2022-03-10:
I don't think that the AS will clean this infomation.

eydonmlau, 2022-03-10:
It's about device session expiration, you can checkout the device_session_ttl value.
[https://forum.chirpstack.io/t/device-address-missing-and-downlink-issue/7886](Refer this)

liangyuanpeng, 2022-03-10:
> It's about device session expiration, you can checkout the device_session_ttl value in network server .toml.

Thanks for your confirm that, i really need to check my production environment for it.

creyD, 2022-10-24:
@zgz757183190 Any update on this? Otherwise this ticket could be closed.

closed 2022-11-01 by brocaar