brocaar/chirpstack-application-server #675: MQTT Integration forces me to use client-certificate?

maxreb, 2022-04-01:
In the newest version when I want to add MQTT Integration on an Application, I only have the possibility to click on "Get Certificate" to generate a client certificate. I don't have the possibility anymore to skip client certificate generation so the client just connects without client certificate. Is this by design? Shall the user not be allowed to connect without an client certificate to another mqtt server? If so, why? 

What I want is to send the application uplink data of an application to an external mqtt server which is not able to handle client certificates. 

[image]

After clicking on "Get Certificate" the text then reads:

> If required by the network, the MQTT client needs to be configured with a client certificate in order to connect to the MQTT broker to device data.
 
In the configuration we have the following comment:

> [application_server.integration.mqtt.client]
> 
>     # CA certificate and key file (optional).
>     #
>     # When setting the CA certificate and key file options, ChirpStack Application Server
>     # will generate client certificates which can be used by the end-application for
>     # authentication and authorization with the MQTT broker.

So either the texts are wrong (since they assume the CA certs are optional) or the normal "add" button is just gone. Or do I misunderstand something here?

https://github.com/brocaar/chirpstack-application-server/blob/65b5d15c55062c583ce9253ba124343e7fa36a80/ui/src/views/applications/integrations/MQTTCard.js#L41

brocaar, 2022-04-04:
MQTT integrations can not be added through the web-interface (this was never possible), only using the `chirpstack-application-server.toml` configuration file. The MQTT integration in the web-interface is just an aid in case you are using client-certificate authentication.

maxreb, 2022-04-04:
Okay, I want to send data from only one application to an external MQTT Server.
For anyone who is curious, I solved it by creating a [mosquitto bridge ](https://medium.com/jungletronics/mosquitto-bridge-5b44e9687fb 3) for the topic of that application.