brocaar/chirpstack-application-server #692: Make GlobalSearch accessible also for authorization with API-Keys

JanZimmer, 2022-07-21:
https://github.com/brocaar/chirpstack-application-server/blob/6ee8de53a83a86f489ada78e2b96ab1b9847e783/internal/api/external/internal.go#L101

As of now, internal API GlobalSearch is not available via API-Key Authorization but should be.

brocaar, 2022-07-21:
It is called internal API for a reason :-) This API is there to facilitate the web-interface, but there is no guarantee that there won't be any breaking changes.

closed 2022-07-21 by brocaar

JanZimmer, 2022-07-21:
@brocaar specifically, I'm using it because, e.g., for the as.application API, if you try to create an app with an already existing name, you get ALREADY_EXISTS Status Code but you have no chance to find Apps by name. Same holds as far as I know for devices... Any ideas on how to circumvent this? Shall I open a ticket to extend the API to be able to look up entities by name also

I think in general this should be made possible for all "unique" properties of all API-accessible entities besides IDs.

brocaar, 2022-07-21:
I think this will be solved by ChirpStack v4, where names no longer have to be unique.