brocaar/chirpstack-application-server #708: Strange data payload after doing unwise restoration

SchroterQuentin, 2022-12-28:
- [ ] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

We had 25 devices connected to our platform and 10 old devices that we no longer needed. My colleague select this old devices clicking on the checkbox from the application page and pressed the delete button in the top right corner... He didn't read the popup and delete the app ... We lost all our devices.

We use chirpstack behind our own application through the API (but we still use chirpstack for debugging). Our own software solution had all the devices information, so we were able to recreate them but on a new app and without the activation keys (all devices are OTAA).
Fortunately, I had made a redis backup the day before so I restored the redis backup and we were able to see all the devices connected again.
But... The data we got from the devices was crap. None of the payloads were correct and we were receiving data that the devices couldn't even send...
I guess there is something that went wrong when I restored the redis but I can't say what exactly. Is there something like a cryptographic key inside the application entity that is used in the device payload decryption process? I'm sure the problem is with the decryption not being done in the right way. And that's not the first time that I restore a chirpstack server but the first time I restore only the redis part.

## What did you expect?

I would expect to have a better colleague...
More seriously, I would expect to have all the communication working again or not working anymore but not having strange payload from the devices. Is there a way to verify that the uncryption keys are correct when receiving a payload ? Like a sumcheck or something like that ?

## Steps to reproduce this issue

Steps:

1. Backup the redis
2. Delete the application entity with your devices
3. Create a new application entity and add your devices (OTAA) without rejoin
4. Restore the redis

## Could you share your log output?

I don't have logs anymore ... It takes us 1 week to see that the data wasn't correct and it takes me 2 days to figure out that the problem wasn't our application (as my colleague tell us) but the data from the devices.

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.17.8  |
| Network Server      | v3.16.5  |
| Gateway Bridge      | v3.14.5 |

brocaar, 2023-01-10:
Please note that in ChirpStack v3, there is a ChirpStack Network Server and ChirpStack Application Server component. Both components have their own databases holding part of the keys. The ChirpStack Network Server validates the MIC using the NwkSKey so in case you restored an older key, the MIC will fail.

For the AppSKey stored in the ChirpStack Application Server database there is no MIC like validation. The encrypted FRMPayload gets decrypted using the AppSKey stored in the database, which will always output data, be it correct or incorrect. It looks like you are getting incorrect data and I expect this is because you might have restored the DB from an older snapshot.

Please note that ChirpStack v4 makes this a lot easier as now both components are merged into one :-)

Hopefully this gives you some hints.

closed 2023-01-10 by brocaar

SchroterQuentin, 2023-01-10:
Thanks for the informations ! At least I have an explanation of what happens.
We will for sure migrate to ChirpStack v4 but we need some time ^^. 
Glad to know it will be easier ! You have already done a very good job ! Thanks a lot 😄